linkease/doc.linkease.com #66: R5S用iStoreOS21.02.3 2022120905，上网测试性能腰斩。

TeddyXiong79, 2022-12-27:
和我的情况一模一样，设备是R5S装载iStoreOS 21.02.3 2022120905。（测试用的同一台电脑，同一根网线）
电脑直连光猫拨号测速：850Mbps左右
用Huawei-Wifi-AX3Pro直连光猫拨号，用有线测速： 840Mbps左右
用Huawei-Wifi-AX3Pro直连光猫拨号，用5Ghz测速：280Mbps左右
用R5S直连光猫拨号（用iStoreOS 21.02.3-2022120905），用有线连接到LAN口测速：550Mbps
用Huawei-Wfi-AX6直连R5S，再由R5S直连光猫拨号，用有线测速：530Mbps
用Huawei-Wifi-AX6直连R5S，再由R5S直连光猫拨号，用5Ghz-Wifi测速：320Mbps
整个全部测试基本上在15分钟内完成。
得出结论：1、R5S用iStoreOS确实速度几乎腰斩。
2、Huawei的AX3和AX6的Wifi性能都是垃圾，请忽略。
3、关注第1个结论。

jjm2473, 2023-01-03:
只测有线就行了，不要用wifi来测。我们没有千兆的外网，没法测试。你测试过R5S的其他固件吗

TeddyXiong79, 2023-01-03:
当然，我只测试有线的速度做为参考（无线仅仅是顺便测试一下的），我测试过Friendly原生的OS（rk3568-sd-friendlywrt-21.02-20221204），测试速度几乎和直连光猫一样的，可以到760Mbps。但用IstoreOS只能测试到550Mbps。

closed 2023-09-02 by jjm2473