linkease/doc.linkease.com #82: istore商店的Jellyfin无法与NasTool联动

Thoams-shenxing, 2023-02-06:
开发者你好：
         NasTool需要Jellyfin的API密钥，然应用商店里提供的版本没有此功能，iStoreOS版本如下图，使用X86主机安装。
         希望Jellyfin镜像可以进行切换，安装时输入拉取地址，而后下载安装。
        有个中国特供版的Jellyfin，nyanmisaka/jellyfin，
镜像是docker pull nyanmisaka/jellyfin:latest，最后附上折腾爱好者的介绍安装链接https://post.m.smzdm.com/p/a3gw6g47/

                 ——iStoreOS爱好使用者慎行敬上

[image]

[image]
[image]
[image]

jjm2473, 2023-02-06:
控制台里面没有吗？

jjm2473, 2023-02-06:
先打开控制台啊，我靠，麻烦让nastools的作者更新下提示文案 https://quickbox.io/knowledge-base/v2/applications/autoscan/create-an-api-key-for-autoscan-in-jellyfin/

jjm2473, 2023-02-06:
nyanmisaka/jellyfin 跟官方配置文件好像不兼容的，切换镜像可能有问题，有需要可以自己用命令部署。

closed 2023-02-06 by jjm2473