linkease/doc.linkease.com #123: Docker容器因文件夹权限丢失导致异常退出

liuwleo, 2023-08-07:
重启istoreOS后，容器会自动推出，发现是文件夹权限问题，重新chmod 777后，可以正常重启。但是之后只要重启istoreOS后仍旧会出现权限丢失，容器退出的问题。每次出现后，手工赋予777权限解决。能否帮忙确认是什么原因。谢谢。Docker已迁移至外挂独立硬盘。

jjm2473, 2023-08-07:
你确定是权限问题？确定不是重启docker就解决了？确定777权限是必要的？

liuwleo, 2023-08-11:
俺想您的建议充气docker，目前没有复现。应该解决了，多谢。

jjm2473, 2023-08-11:
如果是因为docker启动时硬盘还没挂载，那么在后台的杂项设置里面可以配置等待硬盘

closed 2023-08-11 by jjm2473

liuwleo, 2023-08-11:
> 如果是因为docker启动时硬盘还没挂载，那么在后台的杂项设置里面可以配置等待硬盘

请教一下，如何才能确认是因为没有挂载造成的呢？谢谢

jjm2473, 2023-08-11:
如果刚开机docker的容器和镜像都没有，但是重启docker就能恢复容器和镜像，那一般就是因为docker启动时还没挂载好硬盘。准确点可以看看系统日志，docker什么时候启动的，硬盘又是什么时候挂载的。最简单的方式就是配置下等待硬盘，看看之后还会不会出现。

liuwleo, 2023-08-12:
昨晚系统定时重启后，再次出现容器退出，没有自己重启。今天尝试手动重启容器，提示错误如下: Containers: start photoprism...code:400 failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: exec: "/scripts/entrypoint.sh": permission denied: unknown
和之前一样，重新赋予777权限后，成功重启。
请各位高手解惑我，谢谢。

jjm2473, 2023-08-13:
我不是让你配置等待硬盘吗？你有试过吗？

jjm2473, 2023-08-13:
在后台的“系统”-“杂项设置”里面

liuwleo, 2023-08-14:
> 在后台的“系统”-“杂项设置”里面
你好，尝试设置了延迟及等待，但是重启后容器依旧无法正常启动。错误提示和之前一样，解决方案也和以前一样。
[image]

jjm2473, 2023-08-15:
建议重新格式化成ext4试试，或者换个ext4的分区试试

liuwleo, 2023-08-15:
> 建议重新格式化成ext4试试，或者换个ext4的分区试试

目前已经是ext4，见如下截图。如果换分区，docker容器是否需要重装？谢谢。
[image]

jjm2473, 2023-08-15:
我艹，我知道你的文件夹为什么有问题了，你把aria2设置到分区根目录了？然后docker又在里面？

jjm2473, 2023-08-15:
为什么要把docker的数据放在aria2的数据目录里面？这不是傻吗？docker跟aria2有毛线关系啊，你把他们设置成上下级？

liuwleo, 2023-08-15:
我靠，我怎么没发现，好像是错了。那如何进行更正？是修改Docker目录还是

liuwleo, 2023-08-17:
> 还能咋地，aria2和docker都重新配置，要么就恢复出厂设置，把硬盘格式化了，然后再重新配置

重新配置了aria2，问题应该解决，谢谢！