linkease/doc.linkease.com #125: 流量统计设置显示的网卡

dsus4wang, 2023-08-29:
我有两条宽带 wan和wan1，可以指定显示wan1嘛

jjm2473, 2023-08-29:
为啥不配置的时候反过来呢？程序实际上是按默认网关所在的接口作为流量统计的接口。如果你有多个默认网关，那你试试改变配置的顺序

dsus4wang, 2023-08-30:
```
root@iStoreOS:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.214.124.1    0.0.0.0         UG    0      0        0 pppoe-wan
10.214.124.1    *               255.255.255.255 UH    0      0        0 pppoe-wan
172.17.0.0      *               255.255.0.0     U     0      0        0 docker0
192.168.8.0     192.168.8.1     255.255.255.0   UG    0      0        0 eth0
192.168.100.0   *               255.255.255.0   U     0      0        0 br-lan
```
实际上默认网关是pppoe了……仍然没有用

dsus4wang, 2023-08-30:
注：pppoe-wan网卡为eth2，拨号上网，eth0为第二条wan，DHCP获取IP

jjm2473, 2023-08-30:
流量统计的就是pppoe-wan啊

jjm2473, 2023-08-30:
你这个路由表只有一个默认网关，也就是pppoe-wan上的，怎么也不可能统计到其他接口上。就算你说eth0还有dhcp客户端，但是你也没配置默认网关啊

jjm2473, 2023-08-30:
只有一个默认网关的情况下，怎么也不会出错

dsus4wang, 2023-08-30:
[image]
事实上不仅折线图出错了，ip地址的显示出错了 它显示的就是eth0的
相关配置文件的路径是？

jjm2473, 2023-08-30:
现在看看route

dsus4wang, 2023-08-30:
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.214.124.1    0.0.0.0         UG    0      0        0 pppoe-wan
10.214.124.1    *               255.255.255.255 UH    0      0        0 pppoe-wan
172.17.0.0      *               255.255.0.0     U     0      0        0 docker0
192.168.8.0     192.168.8.1     255.255.255.0   UG    0      0        0 eth0
192.168.100.0   *               255.255.255.0   U     0      0        0 br-lan
```
route正常的

jjm2473, 2023-08-30:
你执行下`ubus call network.interface dump`，把整个结果都发过来看看

dsus4wang, 2023-08-30:
```
root@iStoreOS:~# ubus call network.interface dump
{
        "interface": [
                {
                        "interface": "lan",
                        "up": true,
                        "pending": false,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "uptime": 70247,
                        "l3_device": "br-lan",
                        "proto": "static",
                        "device": "br-lan",
                        "updated": [
                                "addresses"
                        ],
                        "metric": 0,
                        "dns_metric": 0,
                        "delegation": true,
                        "ipv4-address": [
                                {
                                        "address": "192.168.100.1",
                                        "mask": 24
                                }
                        ],
                        "ipv6-address": [

                        ],
                        "ipv6-prefix": [

                        ],
                        "ipv6-prefix-assignment": [
                                {
                                        "address": "2409:8a20:835:f760::",
                                        "mask": 60,
                                        "preferred": 102564,
                                        "valid": 188964,
                                        "local-address": {
                                                "address": "2409:8a20:835:f760::1",
                                                "mask": 60
                                        }
                                },
                                {
                                        "address": "fdbc:b22c:f3f6::",
                                        "mask": 60,
                                        "local-address": {
                                                "address": "fdbc:b22c:f3f6::1",
                                                "mask": 60
                                        }
                                }
                        ],
                        "route": [

                        ],
                        "dns-server": [

                        ],
                        "dns-search": [

                        ],
                        "neighbors": [

                        ],
                        "inactive": {
                                "ipv4-address": [

                                ],
                                "ipv6-address": [
[... 346 lines cut ...]
                                "ipv6-address": [

                                ],
                                "route": [

                                ],
                                "dns-server": [

                                ],
                                "dns-search": [

                                ],
                                "neighbors": [

                                ]
                        },
                        "data": {
                                "zone": "wan",
                                "passthru": "001700202409802020000000000000000000000824098020200000000000000000000088",
                                "zone": "wan"
                        }
                }
        ]
}
```

jjm2473, 2023-08-30:
你把vmnet的“使用默认网关”关掉试试

jjm2473, 2023-08-30:
你的配置估计是有问题的，为什么会有
```
                                {
                                        "target": "192.168.8.0",
                                        "mask": 24,
                                        "nexthop": "192.168.8.1",
                                        "source": "0.0.0.0/0"
                                },
```
这样的记录。

你把/etc/config/network的内容发出来，注意消敏

dsus4wang, 2023-08-30:
```
root@iStoreOS:~# cat /etc/config/network 

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdbc:b22c:f3f6::/48'

config interface 'vmnet'
        option device 'eth0'
        option proto 'dhcp'
        option peerdns '0'
        list dns '223.5.5.5'
        list dns '1.1.1.1'

config interface 'vmnet6'
        option device 'eth0'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config device
        option name 'eth2'

config interface 'wan'
        option proto 'pppoe'
        option device 'eth2'
        option username '123'
        option password '123'
        option ipv6 'auto'

config route
        option interface 'wan'
        option target '0.0.0.0/0'

config route
        option interface 'vmnet'
        option target '192.168.8.0/24'
        option gateway '192.168.8.1'
```
最后一条route是用来访问192.168.8.0/24下局域网资源的

jjm2473, 2023-08-30:
```
config route
        option interface 'wan'
        option target '0.0.0.0/0'

config route
        option interface 'vmnet'
        option target '192.168.8.0/24'
        option gateway '192.168.8.1'
```
这两个都完全没用。

wan本身就开了默认网关。

vmnet的网段是192.168.8.0/24，本身就意味着192.168.8.0/24网段的设备都往这个接口发，而且也不可能往192.168.8.1发，因为是同网段的

dsus4wang, 2023-08-30:
第一条是确保默认路由，第二条是因为我现在用8.0/24的网段下的机器来ssh这台机器(192.168.8.46 or 192.168.100.1) 如果删掉会连不上……
但总感觉这两个配置不影响主页显示的吧？

jjm2473, 2023-08-30:
但是你这两个配置以后，在network和route里面不一致啊，
```
                                {
                                        "target": "0.0.0.0",
                                        "mask": 0,
                                        "nexthop": "192.168.8.1",
                                        "source": "192.168.8.46/32"
                                }
```
这个在route里面都没体现出来，原本应该是要么没有这个，而是当作默认网关

jjm2473, 2023-08-30:
这样吧，我在程序里再加个判断条件，对于"source"不是0.0.0.0/0的，就忽略掉

jjm2473, 2023-08-30:
看看会不会影响其他配置的判断

dsus4wang, 2023-08-30:
我把配置删掉了 虽然可以正常上网 但是还是不会正常显示

好的 感谢耐心讲解 麻烦了

jjm2473, 2023-08-30:
但是删掉以后，route应该就会出现多个默认网关了吧

jjm2473, 2023-08-30:
要么你就把vmnet的默认网关关掉，一个系统不可能用两个默认网关的

jjm2473, 2023-08-30:
多wan的情况下，要使用策略路由，不是配置多个默认网关就行了

dsus4wang, 2023-08-30:
> 对啊，那你把vmnet的默认网关功能关掉啊，你又不打算流量从这里出去

懂了 关了就显示正常了 谢谢哥
不过好奇luci这里的默认网关开关是优先级低的吗，我这里开着的时候用route命令查看 并没有相应的默认网关

jjm2473, 2023-08-30:
可能你配置的route冲突了，就没有默认网关了，而且你的场景根本就不需要默认网关，也不需要配置其他路由，一个网段中的ip本来就是直接发送的

dsus4wang, 2023-08-30:
> 可能你配置的route冲突了，就没有默认网关了，而且你的场景根本就不需要默认网关，也不需要配置其他路由，一个网段中的ip本来就是直接发送的

明白了 感谢耐心解答