linkease/doc.linkease.com #139: 无法添加异地互联

love90618, 2023-11-09:
[image]
无法添加异地互联，有大佬可以看看是为啥么？

love90618, 2023-11-09:
[image]

jjm2473, 2023-11-09:
要填一个网段，不是一个ip。例如`192.168.2.0/24`

love90618, 2023-11-09:
[image]
我觉得应该在输入框加一个样例，或者后面加一个标识符解释，要不然用户可能会遇到困惑；

love90618, 2023-11-09:
或者说报错信息能给出提示也可以

jjm2473, 2023-11-09:
你不是要访问一个网段吗？192.168.2.0是一个网段？错误提示CIDR就是指网段，也就是我说的格式。如果你只是为了访问单独一个主机，可以使用上面的“远程设备”功能。

ks-bao, 2023-11-09:
建议很好，感谢。
还有我们的教程在这里：https://doc.linkease.com/zh/guide/linkease/function/remote_connects.html#%E8%BF%9C%E7%A8%8B%E7%BD%91%E6%AE%B5

love90618, 2023-11-09:
> 你不是要访问一个网段吗？192.168.2.0是一个网段？错误提示CIDR就是指网段，也就是我说的格式。如果你只是为了访问单独一个主机，可以使用上面的“远程设备”功能。

这个提示并不是所有人都能看懂的哦，产品面对普通大众使用，在错误提示和解释上，还是需要尽量简单明了一点。

closed 2023-11-09 by love90618