linkease/doc.linkease.com #152: 我的接口是usb3.2 ，插入移动硬盘

jiaz520, 2023-12-22:
我的接口是usb3.2 ，插入移动硬盘，请问是缺少驱动还是因为什么原因。
系统日志：
Fri Dec 22 13:55:57 2023 kern.warn kernel: [10536.189452] usb 2-3-port4: config error
Fri Dec 22 13:55:59 2023 kern.warn kernel: [10538.349561] usb 2-3-port4: config error
Fri Dec 22 13:56:00 2023 kern.warn kernel: [10539.738991] usb 2-3-port4: config error
Fri Dec 22 13:56:03 2023 kern.warn kernel: [10541.908707] usb 2-3-port4: config error
Fri Dec 22 13:56:04 2023 kern.warn kernel: [10543.229050] usb 2-3-port4: config error
Fri Dec 22 13:56:05 2023 kern.warn kernel: [10544.549009] usb 2-3-port4: config error
Fri Dec 22 13:56:07 2023 kern.warn kernel: [10545.899131] usb 2-3-port4: config error
内核日志：
[ 7540.309852] usb 2-3-port4: config error
[ 7541.690525] usb 2-3-port4: config error
[ 7544.000280] usb 2-3-port4: config error
[ 7546.289794] usb 2-3-port4: config error
[ 7548.590143] usb 2-3-port4: config error
[ 7550.759784] usb 2-3-port4: config error
[ 7552.090179] usb 2-3-port4: config error
[ 7553.419682] usb 2-3-port4: config error
[ 7554.739791] usb 2-3-port4: config error
[ 7556.059462] usb 2-3-port4: config error
[ 7558.219758] usb 2-3-port4: config error

jjm2473, 2023-12-22:
什么机器？

jjm2473, 2023-12-22:
如果是istoreos固件问题，麻烦提交到 https://github.com/istoreos/istoreos/issues，如果是梅林固件问题，那就不知道提交到哪了，可能群里反馈

jiaz520, 2023-12-22:
> 什么机器？

就是x86 迷你电脑。3.2接口，京东买的

jjm2473, 2023-12-22:
哪最好咨询下厂商，是不是bios有问题，我这里有R86s，USB 支持 10Gbps，应该也是3.2，没啥问题

jjm2473, 2023-12-22:
你也可以试试新内核的测试版固件 https://fw.koolcenter.com/iStoreOS/alpha/

jiaz520, 2023-12-22:
> 你也可以试试新内核的测试版固件https://fw.koolcenter.com/iStoreOS/alpha/

已经用了最新的固件，R86s usb接口是3.0的。我怀疑3.2接口的usb是不是还不支持。

jjm2473, 2023-12-22:
@jiaz520 测试固件跟正式发布的固件不一样，内核比较新，但是没有在线升级

jjm2473, 2023-12-22:
10Gbps是USB3.0？