linkease/doc.linkease.com #154: frpc 插件设置界面增加复选框

zzonce, 2024-01-03:
frpc 的穿透配置，暂时不用的话只能将设置的服务删除掉。
需要用的时候在增加上，重复操作比较多。
 希望增加复选框，暂时不用的时候可以，先注释一下，暂停使用。

jjm2473, 2024-01-05:
frpc是openwrt官方维护的，需要去openwrt的issue反馈。
其实也不需要卸载重装插件这么麻烦，可以在后台的“系统”-“启动项”里面禁用frpc（点下“已启用”就会切换成“已禁用”），然后点停止，就不会再启动了。

zzonce, 2024-01-05:
您好，看来上边描述的不清楚，让您误会了，官改版本的软件的 UI 是您们维护吗？
我意思是在服务配置处，增加复选框。
取消勾选之后，就相当于在配置文件中，暂时注释不需要的服务配置。
在需要的时候在勾选上，把配置文件中的注释取消，启用这个服务。因为有些服务一直暴漏在公网中的话，担心有被攻击的风险。
[image]

jjm2473, 2024-01-06:
靠，不是istoreos吗？是梅林固件？

zzonce, 2024-01-10:
应该算梅林固件吧
[image]

zzonce, 2024-01-10:
我是不是找错地方了，大佬勿喷😓

jjm2473, 2024-01-10:
倒是没找错地方，不过梅林的开发者估计不会看这个github的，至少我没见过任何梅林相关的开发者回复过。还是群里反馈吧。

zzonce, 2024-01-10:
好的，谢谢答复。