linkease/doc.linkease.com #156: 手机uu加速失败导致软路由无法联网

LeeYouRan, 2024-01-19:
遇到一个奇怪的问题，我的是r2s，下面接了一个路由器，手机上用uu加速器开启加速，有时候会卡在99%，然后提示加速失败，紧接着网络就断了，必须插拔r2s电源才行。但是在电脑上，用uu加速器就没有遇到加速失败的情况，也不会导致断网。

LeeYouRan, 2024-01-19:
一开始以为是插件有影响，因为我也装了uu软路由的插件，以及通过uu主机加速器app安装了软路由插件。但是格式化tf卡重装istore系统，不安装额外的插件，也会遇到这个情况。

jjm2473, 2024-01-23:
那应该就是uu软路由插件的问题，反馈给uu吧，插件是他们提供的

LeeYouRan, 2024-01-23:
一开始我也是这么认为的，所以我重装了istore的系统，格式化盘符重装的，没有安装任何插件，然后手机上用uu加速器加速，失败后依然会断网，我的转路由只负责拨号，其他什么都没有装。

jjm2473, 2024-01-23:
istoreos没装uu也会这样？你说的断网是只有手机断网，还是软路由之下的设备都断网？

LeeYouRan, 2024-01-24:
软路由下所有设备都断网

jjm2473, 2024-01-24:
r2s固件是什么版本

jjm2473, 2024-01-24:
你有ttl连接线吗，把r2s的内核日志提取出来看看。或者你配置下日志的持久化，出现网络问题以后，等待几分钟再重启。

LeeYouRan, 2024-01-24:
root@iStoreOS:~# cat system.log
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain de-sci.org
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain discordapp.net
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain vn.hao123.com
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain seezone.net
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain igotmail.com.tw
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain vital247.org
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain iipdigital.usembassy.gov
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain twittbot.net
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain ienergy1.com
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain pornrapidshare.com
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain business-humanrights.org
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: more servers are defined but not logged
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Jan 24 19:41:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.234 a6:61:1c:4c:37:e4
Wed Jan 24 19:41:03 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.234 a6:61:1c:4c:37:e4
Wed Jan 24 19:41:04 2024 cron.err crond[19301]: crond (busybox 1.35.0) started, log level 5
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]: ttyd 1.6.3 (libwebsockets 4.2.1-09e6c20)
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]: tty configuration:
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:   start command: /bin/login
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:   close signal: SIGHUP (1)
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:   terminal type: xterm-256color
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]: LWS: 4.2.1-09e6c20, loglevel 7
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]: NET CLI SRV H1 H2 WS ConMon IPV6-off
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:    /usr/lib/libwebsockets-evlib_uv.so
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:  ++ [wsi|0|pipe] (1)
Wed Jan 24 19:41:05 2024 daemon.err ttyd[19601]: [2024/01/24 19:41:05:5391] N:  Using foreign event loop...
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:  ++ [vh|0|netlink] (1)
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:  ++ [vh|1|default||7681] (2)
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]: lws_socket_bind: nowsi: source ads 0.0.0.0
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:  ++ [wsi|1|listen|default||7681] (2)
Wed Jan 24 19:41:05 2024 daemon.notice ttyd[19601]:  Listening on port: 7681
Wed Jan 24 19:41:07 2024 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) a6:61:1c:4c:37:e4
Wed Jan 24 19:41:07 2024 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.234 a6:61:1c:4c:37:e4
Wed Jan 24 19:41:08 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.234 a6:61:1c:4c:37:e4
Wed Jan 24 19:41:08 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.234 a6:61:1c:4c:37:e4
Wed Jan 24 19:41:12 2024 daemon.info procd: - init complete -
Wed Jan 24 19:41:34 2024 daemon.notice ttyd[19601]:  ++ [wsisrv|0|adopted] (1)
Wed Jan 24 19:41:34 2024 daemon.notice ttyd[19601]: HTTP / - 192.168.100.135
Wed Jan 24 19:41:34 2024 daemon.notice ttyd[19601]: HTTP /token - 192.168.100.135
Wed Jan 24 19:41:35 2024 daemon.notice ttyd[19601]:  ++ [wsisrv|1|adopted] (2)
Wed Jan 24 19:41:35 2024 daemon.notice ttyd[19601]: WS   /ws - 192.168.100.135, clients: 1
Wed Jan 24 19:41:35 2024 daemon.notice ttyd[19601]: started process, pid: 20580
Wed Jan 24 19:41:39 2024 daemon.err ttyd[19601]: [2024/01/24 19:41:39:9424] N: lws_libuv_closewsi: [wsisrv|0|adopted]
Wed Jan 24 19:41:39 2024 daemon.err ttyd[19601]: [2024/01/24 19:41:39:9428] N: lws_libuv_closewsi: thr 0: [wsisrv|0|adopted] sa left 2: dyn left: 4 (rk 0)
Wed Jan 24 19:41:39 2024 daemon.err ttyd[19601]: [2024/01/24 19:41:39:9432] N:  -- [wsisrv|0|adopted] (1) 5.152s
Wed Jan 24 19:41:41 2024 auth.info login[20580]: root login on 'pts/0'

LeeYouRan, 2024-01-24:
root@iStoreOS:~# cat system.log.old
Thu Jan 21 16:50:47 2016 daemon.err ModemManager[4608]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_NAME=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_ID=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_ID_LIKE=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_VERSION=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_VERSION_ID=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_OS_DETECTION=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_CONTAINER_IS_OFFICIAL_IMAGE=false
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_NAME=iStoreOS
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_ID=istoreos
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_ID_LIKE=lede openwrt
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_VERSION=22.03.6
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_VERSION_ID=22.03.6
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_OS_DETECTION=/etc/os-release
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_HOST_IS_K8S_NODE=false
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_KERNEL_NAME=Linux
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_KERNEL_VERSION=5.10.201
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_ARCHITECTURE=aarch64
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_VIRTUALIZATION=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_VIRT_DETECTION=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CONTAINER=unknown
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CPU_VENDOR=
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CPU_MODEL=
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CPU_FREQ=1296000
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_TOTAL_RAM=1035907072
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Thu Jan 21 16:50:47 2016 daemon.debug ModemManager[4608]: hotplug: cached event found: action=add, name=ttyp7, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyp7
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Thu Jan 21 16:50:47 2016 daemon.debug ModemManager[4608]: hotplug: event reported: action=add, name=ttyp7, subsystem=tty
Thu Jan 21 16:50:47 2016 daemon.err ModemManager[4608]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: SQLite database initialization completed
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Thu Jan 21 16:50:47 2016 daemon.debug ModemManager[4608]: hotplug: cached event found: action=add, name=ttype, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttype
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: Host 'iStoreOS' (at registry as 'iStoreOS') with guid '10c69534-c01c-11e5-b001-e2324a79e12f' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version 'v1.33.1', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Thu Jan 21 16:50:47 2016 daemon.debug ModemManager[4608]: hotplug: event reported: action=add, name=ttype, subsystem=tty
Thu Jan 21 16:50:47 2016 daemon.err ModemManager[4608]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:47 2016 daemon.err netdata[9310]: PROCFILE: Cannot open file '/etc/netdata/.environment'
Thu Jan 21 16:50:47 2016 daemon.debug ModemManager[4608]: hotplug: cached event found: action=add, name=tty0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty0
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: Initializing command server.
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: thread created with task id 10419
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: set name of thread 10419 to PLUGIN[tc]
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: thread created with task id 10417
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: set name of thread 10417 to WEB_SERVER[stat
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: starting worker 2
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: starting worker 3
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: starting worker 4
Thu Jan 21 16:50:47 2016 daemon.info netdata[9310]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
[... cut ...]
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain friendsoftibet.org
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain cosmic.monar.ch
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain churchinhongkong.org
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain londonchinese.ca
Wed Jan 24 19:41:02 2024 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#8053 for domain libertytimes.com.tw

LeeYouRan, 2024-01-24:
istoreOS r2s 22.03.6 2024011910

LeeYouRan, 2024-01-24:
root@iStoreOS:~# cat /root/system.log
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_ARCHITECTURE=aarch64
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_VIRTUALIZATION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_VIRT_DETECTION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CONTAINER=unknown
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_VENDOR=
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_MODEL=
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_FREQ=1296000
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_TOTAL_RAM=1035907072
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=ptypc, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ptypc
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=ptypc, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err ModemManager[4610]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=tty9, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err ModemManager[4610]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: SQLite database initialization completed
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: Host 'iStoreOS' (at registry as 'iStoreOS') with guid '112e86a8-c01c-11e5-aec6-e2324a79e12f' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version 'v1.33.1', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=tty53, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err netdata[9685]: PROCFILE: Cannot open file '/etc/netdata/.environment'
Thu Jan 21 16:50:48 2016 daemon.info pppd[5551]: Remote message: Authentication success,Welcome!
Thu Jan 21 16:50:48 2016 daemon.notice pppd[5551]: PAP authentication succeeded
Thu Jan 21 16:50:48 2016 daemon.notice pppd[5551]: peer from calling number AC:85:3D:D2:FA:2C authorized
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: Initializing command server.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10781
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10781 to EXPORTING
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: CONFIG: cannot load user exporting config '/etc/netdata/exporting.conf'. Will try the stock version.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10782
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10782 to WEB_SERVER[stat
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 2
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 3
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 4
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10783
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10783 to PLUGIN[timex]
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10784
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10784 to PLUGIN[tc]
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10780
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10780 to BACKENDS
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: cleaning up...
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread with task id 10780 finished
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10785
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10785 to PLUGIN[diskspac
[... cut ...]
Wed Jan 24 19:52:57 2024 daemon.notice ttyd[19048]: started process, pid: 19623
Wed Jan 24 19:53:02 2024 daemon.err ttyd[19048]: [2024/01/24 19:53:02:3856] N: lws_libuv_closewsi: [wsisrv|0|adopted]
Wed Jan 24 19:53:02 2024 daemon.err ttyd[19048]: [2024/01/24 19:53:02:3862] N: lws_libuv_closewsi: thr 0: [wsisrv|0|adopted] sa left 2: dyn left: 4 (rk 0)
Wed Jan 24 19:53:02 2024 daemon.err ttyd[19048]: [2024/01/24 19:53:02:3866] N:  -- [wsisrv|0|adopted] (1) 5.112s
Wed Jan 24 19:53:07 2024 auth.info login[19623]: root login on 'pts/0'

LeeYouRan, 2024-01-24:
root@iStoreOS:~# cat /root/system.log.old
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_ARCHITECTURE=aarch64
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_VIRTUALIZATION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_VIRT_DETECTION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CONTAINER=unknown
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_VENDOR=
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_MODEL=
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_FREQ=1296000
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_TOTAL_RAM=1035907072
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=ptypc, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ptypc
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=ptypc, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err ModemManager[4610]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=tty9, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err ModemManager[4610]: hotplug: parent device sysfspath not found
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: SQLite database initialization completed
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: Host 'iStoreOS' (at registry as 'iStoreOS') with guid '112e86a8-c01c-11e5-aec6-e2324a79e12f' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version 'v1.33.1', update every 2, memory mode save, history entries 3996, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Thu Jan 21 16:50:48 2016 daemon.debug ModemManager[4610]: hotplug: event reported: action=add, name=tty53, subsystem=tty
Thu Jan 21 16:50:48 2016 daemon.err netdata[9685]: PROCFILE: Cannot open file '/etc/netdata/.environment'
Thu Jan 21 16:50:48 2016 daemon.info pppd[5551]: Remote message: Authentication success,Welcome!
Thu Jan 21 16:50:48 2016 daemon.notice pppd[5551]: PAP authentication succeeded
Thu Jan 21 16:50:48 2016 daemon.notice pppd[5551]: peer from calling number AC:85:3D:D2:FA:2C authorized
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: Initializing command server.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10781
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10781 to EXPORTING
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: CONFIG: cannot load user exporting config '/etc/netdata/exporting.conf'. Will try the stock version.
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10782
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10782 to WEB_SERVER[stat
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 2
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 3
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: starting worker 4
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10783
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10783 to PLUGIN[timex]
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10784
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10784 to PLUGIN[tc]
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10780
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: set name of thread 10780 to BACKENDS
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: cleaning up...
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread with task id 10780 finished
Thu Jan 21 16:50:48 2016 daemon.info netdata[9685]: thread created with task id 10785
[... cut ...]
Wed Jan 24 19:54:54 2024 daemon.notice ttyd[19048]: WS   /ws - 192.168.100.135, clients: 1
Wed Jan 24 19:54:54 2024 daemon.notice ttyd[19048]: started process, pid: 20201
Wed Jan 24 19:54:59 2024 daemon.err ttyd[19048]: [2024/01/24 19:54:59:6287] N: lws_libuv_closewsi: [wsisrv|2|adopted]
Wed Jan 24 19:54:59 2024 daemon.err ttyd[19048]: [2024/01/24 19:54:59:6291] N: lws_libuv_closewsi: thr 0: [wsisrv|2|adopted] sa left 2: dyn left: 4 (rk 0)
Wed Jan 24 19:54:59 2024 daemon.err ttyd[19048]: [2024/01/24 19:54:59:6294] N:  -- [wsisrv|2|adopted] (1) 5.117s

LeeYouRan, 2024-01-24:
按下面的文档配置的日志持久化，以上是重启以后提取 /root/system.log 和 /root/system.log.old 的文件
https://doc.linkease.com/zh/guide/easepi/question.html#q-%E5%A6%82%E4%BD%95%E8%BF%9B%E8%A1%8C%E6%95%85%E9%9A%9C%E6%8E%92%E6%9F%A5

LeeYouRan, 2024-01-24:
另外，我想问一下，以下命令 uci set system.@system[0].log_file="/root/system.log"
uci commit system
/etc/init.d/log restart
ls /root/system.log 对系统有影响吗，是否会导致日志的文件逐渐变大，是否需要恢复到执行命令之前的状态，如果需要，应该如何恢复。此外，按照文档中的步骤，我在LED配置中增加了heartbeat配置，这个后面是否需要删除。
谢谢。

LeeYouRan, 2024-01-24:
istoreOS r2s 22.03.6 2024011910 固件版本如上

LeeYouRan, 2024-01-26:
更新了最新的istoreOS 固件 22.03.6 2024012613 问题依然存在

jjm2473, 2024-02-01:
你发的日志怎么是同一个呢？下次出问题记一下时间，重启以后马上提取日志，贴日志的时候也说明下出问题的时间点

LeeYouRan, 2024-02-01:
算了，我换系统了，问题可以关闭了。

closed 2024-02-01 by LeeYouRan