linkease/doc.linkease.com #193: 安装后quickstart 没有设置网关ip吗？

sultra, 2024-06-24:
用pve安装后，进 quickstart 就只能设置lan 和子网掩码，那网关呢？如果是客户机和虚拟机是跨网段的怎么版，还要先找一台同网的主机打开它的web界面设置吗？

jjm2473, 2024-06-26:
为啥要在lan设置网关，你的虚拟机是单网口的吗？

sultra, 2024-07-23:
是，就只有一个网口；lan口这只网关有什么问题吗？当旁路由，唯一的网口就是接到交换机和其他设备通讯，其实可以理解是wan口

jjm2473, 2024-07-23:
那你用旁路由向导就可以了

sultra, 2024-07-23:
请问怎么用（怎么打开）旁路由向导？是不是要访问web界面？如果是，当前网口没有设置网关地址，访问的客户端“系统”不在一个子网，访问不到路由器！？我的疑问就，要详细设置就得进WEB界面，然后命令行方式不让设置网关地址，导致非同网段根本访问不到！陷入死循环了！非常困惑！所以命令行内能设置详细的网卡配置吗？

jjm2473, 2024-07-25:
设置好ip和掩码以后，不就可以浏览器访问了吗？

sultra, 2024-08-14:
> 设置好ip和掩码以后，不就可以浏览器访问了吗？

再说一遍，操作端和服务端不在同一个子网内！192.168.7.9/24 去访问 quickstart  的 192.168.33.2/24 ，这能不能理解？

jjm2473, 2024-08-15:
用quickstart命令不就可以设置成同一个网段了吗？

jjm2473, 2024-08-15:
如果你本来就打算用不同网段，你要跨网段访问后台不是只能自定义路由表吗？而且要在网关路由器上增加路由。

sultra, 2024-08-16:
不是不能解决，是明明只要配置下网卡的网关，且这是网卡的最基本配置项，这条路由本来可以自动搞定的，非要绕弯路，服了！简直是“不高兴”啊！

closed 2024-08-16 by sultra