linkease/doc.linkease.com #231: openwrt命令行安装易有云失败

bob329, 2024-12-07:
安装命令：
sh -c "$(curl -sSL http://fw.koolcenter.com/binary/LinkEase/Openwrt/install_linkease.sh)"

日志取报错部分：
//usr/lib/opkg/info/luci-app-linkease.postinst: .: line 2: can't open '/etc/uci-defaults/50_luci-linkease': No such file or directory
Collected errors:
 * pkg_run_script: package "linkease" postinst script returned status 1.
 * opkg_configure: linkease.postinst returned 1.
+ opkg install /tmp/luci-i18n-linkease-zh-cn.ipk
Installing luci-i18n-linkease-zh-cn (git-22.330.29386-e58cb58) to root...
Configuring linkease.
Configuring luci-i18n-linkease-zh-cn.
//usr/lib/opkg/info/luci-i18n-linkease-zh-cn.postinst: .: line 2: can't open '/etc/uci-defaults/luci-i18n-linkease-zh-cn': No such file or directory
Collected errors:
 * pkg_run_script: package "linkease" postinst script returned status 1.
 * opkg_configure: linkease.postinst returned 1.
The linkease version is:
Illegal instruction
Install failed(安装失败)
root@OpenWrt:~#

jjm2473, 2024-12-11:
你这个机器什么架构的？执行` uname -a ` 看看。还有把完整的日志发出来看看

bob329, 2024-12-11:
> 你这个机器什么架构的？执行`uname -a` 看看。还有把完整的日志发出来看看

Linux OpenWrt 5.15.167 #0 SMP Mon Sep 23 12:34:46 2024 armv7l GNU/Linux

bob329, 2024-12-11:
> 感觉你没下载对，我又提供了新的地址：https://fw0.koolcenter.com/binary/LinkEase/alpha/link-ease 或者用：https://fw0.koolcenter.com/binary/LinkEase/alpha/linkease.armv5

这个要咋用来着？

bob329, 2024-12-12:
谢谢，可以用了！

closed 2024-12-12 by bob329