linkease/doc.linkease.com #267: 易有云18.3.2问题

czpeng822, 2025-03-15:
新版的ios的易有云的app 18.3.2中的 照片同步后，再重新检测，视频文件依旧还是显示为不同步？会导致多次重复备份。

czpeng822, 2025-03-17:
[image]
[image]
显示上传成功，但显示备份文件不存在。到硬盘查看文件是存在的。

YGAS, 2025-03-19:
[image]
上传文件名和目标文件名不一致，你这边能提供一个原文件给我测试一下吗？感觉应该跟raw格式相关

czpeng822, 2025-03-19:
[image]
https://github.com/user-attachments/assets/ed0d4e21-8e31-4bb1-b802-9eba9ee3cad7

目前看所有的视频都是这样的。

YGAS, 2025-03-19:
好的，等新版本更新，1.8.15

YGAS, 2025-03-19:
商店更新了，试试修复了吗？

closed 2025-03-26 by YGAS